robotology/event-driven #16: YarpView(YARP QT Image Viewer) don't display the image sequences.

kevinleeex, 2018-03-21:
The iCubGui works properly and Simulated iCub moving ,and yarpdataplayer seems work properly too. However, YarpViews show nothing but black backgroung...I have tried a lot(Two versions of YARP, latest one and last one) but I still got the problem. 
BTW, the dataset I use is the testData_20120803_095402.

Thanks.

arrenglover, 2018-03-21:
Hi @kevinleeex, can you please be more specific of which dataset you are using? I cannot find *testData_20120803_095402* in our published datasets available here: https://figshare.com/s/0abd8f18312bec15b121

kevinleeex, 2018-03-22:
@arrenglover Thanks for your instant reply! I just use the testdata(here:[http://www.yarp.it/group__yarpdataplayerExample.html](http://www.yarp.it/group__yarpdataplayerExample.html)) for checking if the yarpdataplayer works properly.
The OpenCV version that I use is **3.4**.
Thanks again.

arrenglover, 2018-03-22:
Can I clarify that you are trying to use the "event-driven"  libraries, or just YARP in general?

kevinleeex, 2018-03-22:
I was trying to use this "event-driven" library, but the problem is no displaying for Image View, no mater what datasets I use, maybe I should check my YARP environment, I will close this issue and thanks for your replies.

closed 2018-03-22 by kevinleeex