robotology/event-driven #28: event-driven dependencies need to be updated

arrenglover, 2019-09-25:
With the update to event-driven cmake files, external projects that depend on event-driven will also need to be updated. An example of the new cmake can be found here:
https://github.com/robotology/event-driven/blob/cmake_update/documentation/example_module.md

We need to make sure all dependencies will be covered and that the robot will be updated correctly as well.

*important* when we update event-driven we will need to make sure that the repository is correctly cleaned to make sure old libraries are not remaining on the system.

arrenglover, 2019-09-27:
@marco-monforte @Iaxama - can one of you please check the $ROBOT_CODE folder on icublap23 to find all the external libraries that we will have to update?

Iaxama, 2019-09-27:
I'll do it now

arrenglover, 2019-09-27:
Thanks - my plan is to update devel version of event-driven today with the cmake changes. Which means next time a pull is done on event-driven the depending libraries won't compile.

ghoshsuman, 2019-09-27:
I created the folder event-driven-stereo locally on the machine, so that I could get combined stereo output from vPreProcess/AE:o, as well as individual channel outputs vPreProcess/left:o vPreProcess/right:o simultaneously, for my experiments. This functionality could be useful in the main repo. Everything else should be the same.

ghoshsuman, 2019-09-27:
Not sure if it's appropriate to write here, but CONTEXT_DIR in CMakeLists of event-driven is set as "event-driven", whereas inside the source files, while configuring rf contexts "eventdriven" is being used. So, I faced some problems where default .ini config files where not being found. 
@arrenglover

arrenglover, 2019-09-27:
@Iaxama - thanks very much! I'll assign some updates to people. I think event-driven-Denis was for his raster-plot pull request #27 . 

@ghoshsuman - please add a pull request (PR) to event-driven (devel) with your changes. We can discuss the best way to achieve the functionality you need in the PR.

@ghoshsuman -  thanks for mentioning this too, it's part of the update to make sure everything is consistent. It is why I wanted to make sure when we do the udpate that things don't break.

arrenglover, 2019-09-27:
audio_visual_speech/eventLipDetector - @AArriandiaga
attend-track-predict/attention_circle - @Iaxama 
bordercell-attention - @GiuliaDAngelo
capocaccia-2018-VOR - @marco-monforte 
CapoCaccia2019 - @Iaxama (what is this repository? who owns it?)
cc19-headdirnet - @arrenglover 
cne2017_icub - @Iaxama please remove this repository
egomotion_demo - @aikolina 
Event-Driven_Paddleball - @marco-monforte 
particle-filter-tracking - @arrenglover 
solution_event-saccadic-suppression - @Iaxama please remove this one (we need to update the VVV repos, but it might be a bigger job!)
soundTracking - @dgutierrezATC
stereo/coopNetwork - @ghoshsuman 

remember to have a look at this example-module:
https://github.com/robotology/event-driven/blob/cmake_update/documentation/example_module.md
to understand what is needed in the cmake file.

as @ghoshsuman has mentioned - we'll need to update our locations for the config files (rename the folder from eventdriven to event-driven in .local/share/yarp/contexts/)

**important** : when the update goes ahead we will need to remember to remove all the old libraries from iCubContrib_DIR (hopefully a make uninstall works, but we need to check)

please ask if you need help!

marco-monforte, 2019-09-27:
CapoCaccia2019 is an old folder of mines, I'm going to delete it.

ghoshsuman, 2019-09-27:
I updated `stereo/coopNetwork (devel)` as per `cmake_update` in `event-driven` changes

arrenglover, 2019-10-14:
@Iaxama @marco-monforte @GiuliaDAngelo - we need to make sure all the applications running on the robot are working. Today the demo could not run because the repositories were not up-to-date. (@ChiaraBartolozzi). Please tick off the repositories in the top level post once they are done.

closed 2019-12-18 by arrenglover