robotology/event-driven #29: issues with event-driven tutorials

arrenglover, 2019-09-26:
Please leave feedback on tutorials here:

1. mistakes
2. missing steps
3. missing tutorials

aikolina, 2019-09-26:
Personally, I find a bit difficult reading the [README](https://github.com/robotology/event-driven/blob/cmake_update/documentation/README.md) of the documentation because of the many new lines. I understand the intention of highlighting the links to the tutorials, but the line breaking looks un-natural and the links are already sufficiently visible thanks to the different text colouring.

I would prefer something like this, instead:

## Hardware
* Hardware could include sensors: ATIS, event-driven skin, cochlea, or IMU, as well as acquisition boards based on zynq chips: ZCB or z-turn. We'll assume the hardware is correctly connected and powered. However, you may need to [set-up an sd-card for a ZCB or z-turn](howtosetupSD.md).

* If the system should be already ready-to-go, and you just need the board to start acquisition, you can follow these instructions to [connect to the ZCB or z-turn](connect_to_zcb.md).

* Once you have a connection to the board, you need to [set-up the yarpserver](setup_yarpserver.md) to correctly communicate between your laptop and the board.

* Finally you will need to [run `zynqGrabber`](zynqGrabber.md).

* At this point, if you have checked your hardware is streaming events, you can start to read and process them, once your software is correctly installed on the laptop as well.

aikolina, 2019-09-26:
There is a styling issue in line 27 of [1viewer.md](https://github.com/robotology/event-driven/blob/cmake_update/documentation/1viewer.md).

Instead of using [blockquotes](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#blockquotes):
> yarp namespace /<yourname>

It would be better to use [code blocks](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code-and-syntax-highlighting):
```bash
yarp namespace /<yourname>
```

In this way, code is nicely highlighted (not YARP related keywords, tough) and special symbols should not disappear due to markdown styling.

lsrosa, 2019-09-26:
On the **Testing your install with the viewer** tutorial [1viewer.md](https://github.com/robotology/event-driven/blob/cmake_update/documentation/1viewer.md).

The first commands seem to be for both testing with the dataset and camera: 
    download the sample dataset from here and unpack to a location of your choosing.
    set the yarp namespace
    `yarp namespace /`
    run a yarpserver
    `yarpserver --write`

These commands could be moved to the section **Using actual hardware/camera? Skip ahead. Using a dataset? Follow these instructions:**

aikolina, 2019-09-26:
For easy changes (e.g. adding link, typos, etc.), I would suggest to make a new branch with the suggested changes and see the difference with respect to `cmake_update` branch by making a PR. In this way, as long as someone else checks the PR, half of the work is already done!

I tried to do so for [1viewer.md](https://github.com/robotology/event-driven/blob/cmake_update/documentation/1viewer.md), see PR #30.

What do you think about it @arrenglover?

lsrosa, 2019-09-26:
On **Writing an example module for event-driven** tutorial [example_module.md](https://github.com/robotology/event-driven/blob/cmake_update/documentation/example_module.md)

While personalizing the project, a command to change the name of the **example-module.ini** file is missing:
`mv example-module.ini <my-module-name>.ini`

arrenglover, 2019-09-26:
> For easy changes (e.g. adding link, typos, etc.), I would suggest to make a new branch with the suggested changes and see the difference with respect to `cmake_update` branch by making a PR. In this [...]

It's a good idea. feel free to put all changes into a PR. I would like to perform the update to devel tomorrow.

aikolina, 2019-09-26:
> While personalizing the project, commands to change the name of the **example-module.ini** and **app_example-module.xml** are missing:
> `mv example-module.ini <my-module-name>.ini`
> `mv app_example-module.xml app_<my-module-name>.xml`

@lsrosa I think I applied the changes you suggested in PR #30, could you please double check?

lsrosa, 2019-09-27:
> @lsrosa I think I applied the changes you suggested in PR #30, could you please double check?

@aikolina great. However, maybe we can change the three lines 

>> `mv <my-module-name>/example-module.cpp <my-module-name>/<my-module-name>.cpp`
>> `mv example-module.ini <my-module-name>.ini`
>> `mv app_example-module.xml app_<my-module-name>.xml`

to
`rename '~s/example-module/<my-module-name>/g' *`, if `rename` is not installed: `sudo apt-get install rename`

We can suggest to `sed` the lines on the CMakelists and .cpp files as well, but `sed` is dangerous (regex are not reversible) and I think that is good for the user/developer to spot where these definitions are in the code.

lsrosa, 2019-09-27:
On the [Writing an example module for event-driven](https://github.com/robotology/event-driven/blob/cmake_update/documentation/example_module.md) tutorial

There are some definitions in the **example-module** that were not updated to **<my-module-name>**, resulting in both names mixed in the yarpmanager. It follows the extra necessary modifications:

1. Add line 24 to the list of lines to be changed on **<my-module-name>.cpp** file
> On line 7, 19, and 124 change the class, constructor and declaration to <my-module-name>.
> ctrl+o, ctrl+x

2. Add to change the first line of file <my-module-name>.ini
`name /example-module` → `name /<my-modulo-name>`

3. Add to change all entries of file app_<my-module-name>.xml, here the sed command will be useful
`sed -n  's/example-module/<my-modulo-name>/g' <my-modulo-name>.xml`

After this, the entries in the yarpmanager display the correct name (`<my-module-name>`)

aikolina, 2019-09-27:
@lsrosa Can you make all the changes you suggested to the [Writing an example module for event-driven](https://github.com/robotology/event-driven/blob/cmake_update/documentation/example_module.md) tutorial on the [PR](#30) I'm working on too? Thanks!

arrenglover, 2019-09-27:
Thanks @aikolina  and @lsrosa. I've pulled in your changes up to here. I made a change to the install process. Can someone check the new version works?

closed 2019-10-28 by arrenglover