robotology/event-driven #47: undistortion remapping error

ghoshsuman, 2019-11-08:
Problems occurring when `undistort==true` for vPreProcess module:
* For stereo-rectification and undistortion, all event pixels get remapped to (0, 0).  

Dense shared projection: [image]
Sparse shared projection:[image]

* When [STEREO_DISPARITY] params are removed from **stefi_calib.ini** (i.e. doing only undistortion), vFramer (AE draw) crashes with segmentation fault. This is probably because vPreProcess remaps event pixels (using instrinsic parameters) in the output space of (608, 480) whereas vFramer expects events to be in (304, 240) space.

Sparse shared projection:[image]
Sparse shared projection:[image]

I'm using input data and camera parameters from the ATIS cameras of the iCub robot, where the calibration was done using asymmetric grid of circles. The last time I saw it working using the same data and params was immediately before merging cmake_update changes into devel branch. There was some distortion at the corners due to poor calibration, but events were getting properly rectified nevertheless.

arrenglover, 2019-11-11:
it seems like not all events are being mapped to 0 but most of them. The projected stereo looks terrible - i.e. only using a couple of hundred pixels between both the images. To me it seems like a poor calibration rather than a bug. I wonder if you can check the rectification results using the old method (i.e. roll back to before the new method was introduced?)

ghoshsuman, 2019-11-25:
The current code seems to be working properly with the `undistort` flag. The calibration parameters I used above were faulty. A bug in the old code related to assignment of the image cv::Size vector during rectification transformation estimation somehow made those faulty parameters "work". Closing the issue.

closed 2019-11-25 by ghoshsuman