robotology/event-driven #60: event-driven in superbuild

arrenglover, 2019-12-18:
Any changes required to integrate event-driven into the superbuild need to be understood and updated.

traversaro, 2019-12-18:
Hi @arrenglover, if you need any help on this feel free to ping me.

arrenglover, 2021-01-27:
https://github.com/robotology/robotology-superbuild/pull/502

closed 2021-01-27 by arrenglover

traversaro, 2021-01-27:
Hi @arrenglover , if you could also do a recent release tag for event-driven so we use that in place of a fixed commit as done in https://github.com/robotology/robotology-superbuild/blob/master/releases/latest.releases.yaml#L173 it would be great.

arrenglover, 2021-01-29:
FYI @traversaro I have made a new release tag v1.5 - https://github.com/robotology/event-driven/releases/tag/v1.5

traversaro, 2021-01-29:
Thanks @arrenglover !