robotology/event-driven #66: `exportIitYarp` generates YARP bottles with badly formatted bottle time-stamp

aikolina, 2020-03-09:
## Description

The timestamps of the bottles generated by [`exportIitYarp`](https://github.com/robotology/event-driven/blob/python_dev/python/libraries/bimvee/exportIitYarp.py) seem badly formatted. Here is an example:

```
0 7 288000000.11631 AE (911 4726831 1002 4784726 ...)
```

What I would expect is something like:

```
BOTTLE_ID YARP_TS AE (911 4726831 1002 4784726 ...)
```

However, I got an extra number before the `AE` tag which I was not expecting.

## Details

Here are the steps I followed:

1. I dumped the raw data using the `binary-dumper`.

2. I imported the raw data using:

   ```python
   imported = importIitYarp(filePathOrName = filePathOrName, tsBits = 30, convertSamplesToImu=False)
   ```

   and checked the time-stamp value for the first event:

   ```pyhont
   imported['data']['right']['dvs']['ts'][0] = 7.28800000e-05
   ```

3. I exported the imported raw data using:

   ```python
   exportIitYarp(imported, exportFilePath= os.path.join(prefix, dataset_root, dataset_name + "_converted"), pathForPlayback= os.path.join(prefix, dataset_root, dataset_name + "_converted"), dataTypes = ['sample', 'dvs'], overwrite=True)
   ```

4. I checked the exported `data.log` and noticed that the bottle envelope was badly formatted.

   ```
   0 7 288000000.11631 AE (911 4726831 1002 4784726 ...)
   ```

So, it seems that the time-stamp in the exponential format (see point 2.) is exported as two separate numbers (see point 4.), and this causes an extra value before the `AE` tag.

aikolina, 2020-03-09:
CC @simbamford

simbamford, 2020-03-10:
Fixed in commit a5c9b4fa

closed 2020-03-10 by simbamford

aikolina, 2020-03-10:
Thanks @simbamford! Everything working as expected now. :)