robotology/event-driven #77: Guarantee stable versions on `master` branch

aikolina, 2020-03-27:
Since we got rid of the `devel` branch, I think we should guarantee stable versions of `master`.

For that, we should probably start to tag the `master` branch more often and plan the releases.

Any other suggestion is more than welcome!

arrenglover, 2020-04-09:
Tags and releases should be handled differently for the way in which the project is used. YARP for example has many users who typically just want to use it and don't want to fix bugs in it - it needs releases. Typically EDPR is the only one that uses event-driven, and we are the ones that should all be on master making sure all the bugs get fixed.

Therefore, at the moment a release schedule etc. isn't high priority. At the moment I tag it when there is any major updates. If we break backwards compatibility I would move the version to 2.0.

aikolina, 2020-04-10:
I understood your point @arrenglover, and I agree with the fact that release plans is not crucial for this repository.

closed 2020-04-10 by aikolina