robotology/event-driven #87: align IMU axis to image reference frame

arrenglover, 2020-04-09:
It will save us some headache to align the IMU sensor as closely as possible to the image plane reference frame.

[image]

**Note: there will always be a transform between the image plane and the IMU which should be taken into account for any precise estimations!**

Given the way the IMU is mounted I calculate the following axes to align the IMU and camera reference frames:

| sensor # | axis |
|-- | --|
| 0 | -a_y |
| 1 | a_x |
| 2 | a_z |
| 3 | -w_y |
| 4 | w_x |
| 5 | w_z |
| 6 | temp |
| 7 | -m_y |
| 8 | m_x |
| 9 | m_z |

The magnetometer I didn't measure, but I assume follows the same pattern as the accelerometer and gyro.

To integrate this axis frame into event-driven I plan to do the following for the C++ side:
1. make an enumeration that follows the above pattern that can be used to for sensor comparison without having to know the actual values (e.g. if IMUevent.sensor == AXIS_Y). This will allow the "swapping" of axis between x and y.
2. perform the negation of the y axis in the covert_to_si function. This function is currently just local to code that currently uses the IMU but can be moved into the library.

@simbamford will handle the python side of things. Do you have any doubts? Is it easy for you to integrate this rotation for the IMU?

simbamford, 2020-04-09:
Great - no problem. I've just put the required changes in branch: stefiImuOrientation

arrenglover, 2020-04-09:
FYI: you can link pull request like this: #88 (this is mine for the same issue)

closed 2020-05-11 by arrenglover