robotology/event-driven #111: Compilation errors with newer GCC version

ddetommaso, 2020-06-15:
Branch: opencv4
GCC: gcc (Ubuntu 9.3.0-10ubuntu2) 9.3.0
CMake: cmake version 3.16.3

`Scanning dependencies of target event-driven
[  9%] Building CXX object lib/CMakeFiles/event-driven.dir/src/codecs/codec_AddressEvent.cpp.o
[ 14%] Building CXX object lib/CMakeFiles/event-driven.dir/src/vCodec.cpp.o
[ 19%] Building CXX object lib/CMakeFiles/event-driven.dir/src/vPort.cpp.o
[  9%] Building CXX object lib/CMakeFiles/event-driven.dir/src/vtsHelper.cpp.o
[ 28%] Building CXX object lib/CMakeFiles/event-driven.dir/src/codecs/codec_GaussianAE.cpp.o
[ 28%] Building CXX object lib/CMakeFiles/event-driven.dir/src/codecs/codec_FlowEvent.cpp.o
[ 33%] Building CXX object lib/CMakeFiles/event-driven.dir/src/codecs/codec_IMUEvent.cpp.o
[ 38%] Building CXX object lib/CMakeFiles/event-driven.dir/src/codecs/codec_LabelledAE.cpp.o
[ 42%] Building CXX object lib/CMakeFiles/event-driven.dir/src/codecs/codec_SkinEvent.cpp.o
[ 47%] Building CXX object lib/CMakeFiles/event-driven.dir/src/codecs/codec_SkinSample.cpp.o
[ 52%] Building CXX object lib/CMakeFiles/event-driven.dir/src/codecs/codec_vEvent.cpp.o
[ 57%] Building CXX object lib/CMakeFiles/event-driven.dir/src/vIPT.cpp.o
[ 61%] Building CXX object lib/CMakeFiles/event-driven.dir/src/vDraw_basic.cpp.o
[ 66%] Building CXX object lib/CMakeFiles/event-driven.dir/src/vDraw_ISO.cpp.o
[ 71%] Building CXX object lib/CMakeFiles/event-driven.dir/src/vDraw_skin.cpp.o
/home/docky/event-driven/lib/src/vDraw_basic.cpp: In member function 'virtual void ev::imuDraw::draw(cv::Mat&, const vQueue&, int)':
/home/docky/event-driven/lib/src/vDraw_basic.cpp:264:25: error: jump to case label
  264 |         case imuHelper::GYR_Y: {//rot y
      |                         ^~~~~
/home/docky/event-driven/lib/src/vDraw_basic.cpp:262:18: note:   crosses initialization of 'cv::Point_<int> pos'
  262 |             auto pos = centre + cv::Point(Ylimit * 0.25, 0);
      |                  ^~~
/home/docky/event-driven/lib/src/vDraw_basic.cpp:268:25: error: jump to case label
  268 |         case imuHelper::GYR_X: {//rot x
      |                         ^~~~~
/home/docky/event-driven/lib/src/vDraw_basic.cpp:262:18: note:   crosses initialization of 'cv::Point_<int> pos'
  262 |             auto pos = centre + cv::Point(Ylimit * 0.25, 0);
      |                  ^~~
/home/docky/event-driven/lib/src/vDraw_basic.cpp:272:25: error: jump to case label
  272 |         case imuHelper::GYR_Z: {//rot z
      |                         ^~~~~
/home/docky/event-driven/lib/src/vDraw_basic.cpp:262:18: note:   crosses initialization of 'cv::Point_<int> pos'
  262 |             auto pos = centre + cv::Point(Ylimit * 0.25, 0);
      |                  ^~~
/home/docky/event-driven/lib/src/vDraw_basic.cpp:276:25: error: jump to case label
  276 |         case imuHelper::TEMP: {//temp
      |                         ^~~~
/home/docky/event-driven/lib/src/vDraw_basic.cpp:262:18: note:   crosses initialization of 'cv::Point_<int> pos'
  262 |             auto pos = centre + cv::Point(Ylimit * 0.25, 0);
      |                  ^~~
/home/docky/event-driven/lib/src/vDraw_basic.cpp:280:25: error: jump to case label
  280 |         case imuHelper::MAG_Y: {//mag y
      |                         ^~~~~
/home/docky/event-driven/lib/src/vDraw_basic.cpp:262:18: note:   crosses initialization of 'cv::Point_<int> pos'
  262 |             auto pos = centre + cv::Point(Ylimit * 0.25, 0);
      |                  ^~~
/home/docky/event-driven/lib/src/vDraw_basic.cpp:282:25: error: jump to case label
  282 |         case imuHelper::MAG_X: {//mag x
      |                         ^~~~~
/home/docky/event-driven/lib/src/vDraw_basic.cpp:262:18: note:   crosses initialization of 'cv::Point_<int> pos'
  262 |             auto pos = centre + cv::Point(Ylimit * 0.25, 0);
      |                  ^~~
/home/docky/event-driven/lib/src/vDraw_basic.cpp:284:25: error: jump to case label
  284 |         case imuHelper::MAG_Z: {//mag z
      |                         ^~~~~
/home/docky/event-driven/lib/src/vDraw_basic.cpp:262:18: note:   crosses initialization of 'cv::Point_<int> pos'
  262 |             auto pos = centre + cv::Point(Ylimit * 0.25, 0);
      |                  ^~~
/home/docky/event-driven/lib/src/vDraw_basic.cpp:294:40: error: qualified-id in declaration before '=' token
  294 | const std::string rasterDraw::drawtype = "RASTER";
      |                                        ^
/home/docky/event-driven/lib/src/vDraw_basic.cpp:296:36: error: qualified-id in declaration before '(' token
  296 | std::string rasterDraw::getDrawType()
      |                                    ^
/home/docky/event-driven/lib/src/vDraw_basic.cpp:301:37: error: qualified-id in declaration before '(' token
  301 | std::string rasterDraw::getEventType()
      |                                     ^
/home/docky/event-driven/lib/src/vDraw_basic.cpp:306:22: error: qualified-id in declaration before '(' token
  306 | void rasterDraw::draw(cv::Mat &image, const ev::vQueue &eSet, int vTime)
      |                      ^
/home/docky/event-driven/lib/src/vDraw_basic.cpp:335:41: error: qualified-id in declaration before '=' token
  335 | const std::string cochleaDraw::drawtype = "EAR";
      |                                         ^
/home/docky/event-driven/lib/src/vDraw_basic.cpp:337:37: error: qualified-id in declaration before '(' token
  337 | std::string cochleaDraw::getDrawType()
      |                                     ^
/home/docky/event-driven/lib/src/vDraw_basic.cpp:342:38: error: qualified-id in declaration before '(' token
  342 | std::string cochleaDraw::getEventType()
      |                                      ^
/home/docky/event-driven/lib/src/vDraw_basic.cpp:347:23: error: qualified-id in declaration before '(' token
  347 | void cochleaDraw::draw(cv::Mat &image, const ev::vQueue &eSet, int vTime)
      |                       ^
/home/docky/event-driven/lib/src/vDraw_basic.cpp:380:42: error: qualified-id in declaration before '=' token
  380 | const std::string interestDraw::drawtype = "AE-INT";
      |                                          ^
/home/docky/event-driven/lib/src/vDraw_basic.cpp:382:38: error: qualified-id in declaration before '(' token
  382 | std::string interestDraw::getDrawType()
      |                                      ^
/home/docky/event-driven/lib/src/vDraw_basic.cpp:387:39: error: qualified-id in declaration before '(' token
  387 | std::string interestDraw::getEventType()
      |                                       ^
/home/docky/event-driven/lib/src/vDraw_basic.cpp:392:24: error: qualified-id in declaration before '(' token
  392 | void interestDraw::draw(cv::Mat &image, const ev::vQueue &eSet, int vTime)
      |                        ^
/home/docky/event-driven/lib/src/vDraw_basic.cpp:439:41: error: qualified-id in declaration before '=' token
  439 | const std::string clusterDraw::drawtype = "CLE";
      |                                         ^
/home/docky/event-driven/lib/src/vDraw_basic.cpp:441:37: error: qualified-id in declaration before '(' token
  441 | std::string clusterDraw::getDrawType()
      |                                     ^
/home/docky/event-driven/lib/src/vDraw_basic.cpp:446:38: error: qualified-id in declaration before '(' token
  446 | std::string clusterDraw::getEventType()
      |                                      ^
/home/docky/event-driven/lib/src/vDraw_basic.cpp:451:23: error: qualified-id in declaration before '(' token
  451 | void clusterDraw::draw(cv::Mat &image, const vQueue &eSet, int vTime)
      |                       ^
/home/docky/event-driven/lib/src/vDraw_basic.cpp: At global scope:
/home/docky/event-driven/lib/src/vDraw_basic.cpp:500:1: error: expected '}' at end of input
  500 | } //namespace ev::
      | ^
/home/docky/event-driven/lib/src/vDraw_basic.cpp:22:14: note: to match this '{'
   22 | namespace ev {
      |              ^
make[2]: *** [lib/CMakeFiles/event-driven.dir/build.make:219: lib/CMakeFiles/event-driven.dir/src/vDraw_basic.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:239: lib/CMakeFiles/event-driven.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
`

arrenglover, 2020-06-17:
@ddetommaso I pushed an update to the branch that fixed these issues for me. I didn't have time to check it runs correctly. For sure this breaks the backwards compatibility with opencv3. let me know if it is fixed also for you.

ddetommaso, 2020-06-17:
Hi @arrenglover,

it compiles perfectly! Thanks

closed 2020-06-17 by ddetommaso