robotology/event-driven #138: Compilation broken with OpenCV 3.2.0 provided by Ubuntu 18.04 and Debian Buster (current Stable)

traversaro, 2021-05-04:
Hi @arrenglover @lunagava,  
   the PR https://github.com/robotology/event-driven/pull/136 broke compilation against the OpenCV version in Ubuntu 18.04, that fails with: 
~~~
[ 51%] Building CXX object lib/CMakeFiles/event-driven.dir/src/vIPT.cpp.o
/home/runner/work/robotology-superbuild/robotology-superbuild/src/event-driven/lib/src/vIPT.cpp: In member function ‘void ev::vIPT::printValidCalibrationValues()’:
/home/runner/work/robotology-superbuild/robotology-superbuild/src/event-driven/lib/src/vIPT.cpp:365:25: error: ‘cv::Size {aka class cv::Size_<int>}’ has no member named ‘empty’
         if(!size_cam[i].empty()) {
                         ^~~~~
lib/CMakeFiles/event-driven.dir/build.make:257: recipe for target 'lib/CMakeFiles/event-driven.dir/src/vIPT.cpp.o' failed
make[5]: *** [lib/CMakeFiles/event-driven.dir/src/vIPT.cpp.o] Error 1
~~~

See https://github.com/robotology/robotology-superbuild/runs/2497106466?check_suite_focus=true . 

The reason is that `cv::Size_::empty`  is not available in the ancient version of OpenCV intalled via apt for Ubuntu 18.04 and Debian Buster. I think we have two alternatives:
* Drop compatibility in event-driven for Ubuntu 18.04 and Debian Buster. If you are not using Ubuntu 18.04 or Debian Buster anymore, we can simply drop Ubuntu 18.04 compatibility for event-driven and for the ROBOTOLOGY_USES_EVENT_DRIVEN option, as we are already doing for other superbuild profiles (see https://github.com/robotology/robotology-superbuild/issues/481) 
* Otherwise, if you care for Ubuntu 18.04 or Debian Buster compatibility, we can fix the code (see https://github.com/opencv/opencv/blob/68d15fc62edad980f1ffa15ee478438335f39cc3/modules/core/include/opencv2/core/types.hpp#L1738)

arrenglover, 2021-05-04:
thanks @traversaro. fixed with #139

closed 2021-05-04 by arrenglover

traversaro, 2021-05-04:
Thanks @arrenglover !