robotology/event-driven #163: Cleanup use of YARP deprecated classes (strikes back)

traversaro, 2022-05-30:
Similar to https://github.com/robotology/event-driven/issues/155, https://github.com/robotology/yarp/pull/2846 removed several YARP classes that had been deprecated for a long time:
* `yarp::os::RateThread`, example migration: https://github.com/robotology/icub-main/pull/525
* `yarp::os::Mutex`, example migration: https://github.com/robotology/icub-main/pull/618
* `yarp::os::LockGuard`, example migration: https://github.com/robotology/icub-main/pull/618

So currently this repo does not compile with YARP devel due to its use of RateThread .

arrenglover, 2022-05-31:
:laughing: thanks again @traversaro, we'll try to fix this as soon as possible.

traversaro, 2022-05-31:
Thanks @arrenglover ! Fortunatly YARP 3.7 was released before, so probably this change will be released on YARP 3.8 in some time.

traversaro, 2023-03-02:
Thanks @arrenglover ! I think the YARP 3.8 is planned in the next days, and we plan to do a distro release after the 6th of March.