robotology/event-driven #164: Discuss how to combine ev2-dev software with stereoCalib from icub-main

xEnVrE, 2022-06-06:
In this issue I will report some issues me and @LZC-68 experienced while trying to run the calibration between an RGB camera, namely a RealSense camera, and an ATIS Generation 3 event camera.

We found the following inconsistencies that we would like to understand how to fix, if possible.

#### Timestamps

The `stereoCalib` module is [checking and comparing the timestamps](https://github.com/robotology/icub-main/blob/4bc196a80016689853714b31b14018996c8cc615/src/tools/stereoCalib/src/stereoCalibThread.cpp#L166) between the two sources (left and right cameras) for each pair of successfully received images. If the difference in timestamp is too much, beyond a given threshold, the images are skipped. The threshold is hardcoded (check [here](https://github.com/robotology/icub-main/blob/4bc196a80016689853714b31b14018996c8cc615/src/tools/stereoCalib/include/stereoCalibThread.h#L94)).

At the moment, the Realsense device is adding its timestamp to the sent image as can be seen [here](https://github.com/robotology/yarp-device-realsense2/blob/77dd21079933c68187da0dd4e6d5e93973e61565/src/devices/realsense2/realsense2Driver.cpp#L1091). It is using the `YARP` `Stamp` class to do so. The time seems to be updated using the `update` method (which probably relies on the `YARP` clock). The stamp is then sent within the envelope of the image transmitted over the port.

On the contrary, the `vFramer` is not sending any stamp:

https://github.com/robotology/event-driven/blob/7bfd331318ddda21bea2a1de662bf8528d45b105/cpp_tools/vFramer/drawers.cpp#L33-L47

We could adapt to the way the Realsense is sending stamps. However, it would not work if we then want to calibrate the event camera with yet another camera (either RGB or event-based) that is sending timestamps using a different mechanism. I guess the problem will remain anyways if the cameras are running on the same `YARP` network but on two different machines, hence with different clocks. Probably, this check was designed having in mind two iCub cameras whose output is generated by the very same module on the same machine, i.e. the icub head machine.

As a workaround, and in order to proceed with the calibration, we are just commenting the line which checks the timestamp (as we are sure the synchronization is good enough per se).

#### PixelMono vs PixelRGB

The `stereoCalib` module is expecting images to be received via two `yarp::os::BufferedPort<yarp::sig::Image<yarp::sig::PixelRGB>>` ports. However, for the calibration we are using the `vFramer` with the `black` image mode as output. This produces images with 1 unsigned channel

https://github.com/robotology/event-driven/blob/7bfd331318ddda21bea2a1de662bf8528d45b105/cpp_tools/vFramer/drawers.cpp#L149

that are converted back to `yarp::sig::Image<yarp::sig::PixelMono>` via the `YARP/OpenCV` interoperability methods

https://github.com/robotology/event-driven/blob/7bfd331318ddda21bea2a1de662bf8528d45b105/cpp_tools/vFramer/drawers.cpp#L37-L40

before being sent.

We don't know the effect of this mismatch. At the moment, in order to proceed with the calibration, we modified the code from:

https://github.com/robotology/event-driven/blob/7bfd331318ddda21bea2a1de662bf8528d45b105/cpp_tools/vFramer/drawers.cpp#L144-L157
.
to

```
// BLACK DRAW //
// =========== //
void blackDrawer::updateImage()
{
    if(canvas.empty())
        canvas = cv::Mat(img_size, CV_8UC3);
    else
        canvas = 0;

    ev::info inf = input.readSlidingWinT(window_size, false);

    for (auto &v : input)
    {
	cv::Vec3b& color = canvas.at<cv::Vec3b>(v.y, v.x);
	color[0] = 255;
	color[1] = 255;
	color[2] = 255;
    }
}
```

We would like to know your opinion on these two issues.

cc @arrenglover @Iaxama 

cc @LZC-68

arrenglover, 2022-06-06:
1. timestamps need to be added to the image outputs from the vFramer
2. black draw should output 3-channel images to be compatible with the calibration
I'll try to fix asap

arrenglover, 2022-06-09:
@xEnVrE @LZC-68 - these issues should be fixed in latest push. let me know.

closed 2023-03-07 by arrenglover